GilesBathgate/RapCAD #10: The menu to move the camera do nothing

marcoKoH2, 2015-11-25:
in ubuntu version 0.9.7~trusty1
the entries menu View -> Top Bottom Left Right Front Back and shortcuts Ctrl+[4-9]
do nothing.

GilesBathgate, 2015-11-25:
fixed in 6b7b194e9eeb873c0a2b98c386527a3a6457092b

closed 2015-11-25 by GilesBathgate

marcoKoH2, 2015-11-25:
thanks very fast !
I've to compile of my (I've middle knowledges)
or I can wait (1-2 week)
a new release on your repo ?
thanks again

GilesBathgate, 2015-11-25:
You can use a snapshot version (rapcad - 0.9.7+1SNAPSHOT20151125~trusty1) by adding ppa:gilesbathgate/rapcad-snapshot to your system's Software Sources.

marcoKoH2, 2015-11-25:
I've just finish to compile to myself using QtCreator (rapcad.pro)
all works fine, thanks again.

GilesBathgate, 2015-12-01:
(Since github have removed the private messaging feature I will post here, which is not ideal)

How did you get on with RapCAD in general, were you able to do what you wanted? Did you find any other missing features or bugs that you really want fixed, or implemented?