GilesBathgate/RapCAD #11: Instructions on building for Mac OS X

juliendorra, 2016-01-28:
Hi, 
The last Mac release seems very outdated (0.9.0).

Of course a new, up to date, pre-build release for OS X would be great (for us Mac users :-) but building instructions could also be useful and would allow knowledgable users to help and contribute the build instead of @GilesBathgate having to do it for each release, which may or may be impractical for you.

GilesBathgate, 2016-01-28:
@juliendorra Ideally I would like automated builds for mac-osx, but you are right additional (and up to date) build instructions for macosx would be good. The [current instructions](https://github.com/GilesBathgate/RapCAD/blob/master/doc/compiling.asciidoc#mac-os-x-1062) are a bit bare.

I don't have a very large user base, but its interesting to hear that at least a few macosx users are interested.

GilesBathgate, 2017-11-26:
@juliendorra I currently have build issues with mac-osx and I am not sure how to fix it:

https://travis-ci.org/GilesBathgate/RapCAD/jobs/305545797

Until I can figure that out I will regret to say that mac-osx support is unavailable.

GilesBathgate, 2017-11-26:
@juliendorra I have managed to fix the build issue. So at least I know that RapCAD does build on mac-osx.  All that is outstanding now is to write some build instructions as originally requested.

For the record, the compilation error was due to parser.y defining tokens IF and AS that conflicted with headers included from CGAL.

https://travis-ci.org/GilesBathgate/RapCAD/jobs/307509461

GilesBathgate, 2020-01-21:
> Would it make sense to generate all the objects in a sub directory?
Typically you would do a "shadow" build which means changing to a build dirrectory and issuing:

`/usr/lib/qt5/bin/qmake -o Makefile ../rapcad/rapcad.pro -spec linux-clang`

ajmas, 2020-01-21:
Okay, I'll look to updating the PR.

Also, WRT to `qmake` path, is looks like it isn't in the search path by default: https://stackoverflow.com/questions/31315354/qt5-and-macports-missing-links/31318672#31318672

commit dec13ec6 (2020-01-31): Fixes #11 Updated docs for macOS (#16)

* Fixes #11 Updated docs for macOS
* Use build directory for artefacts
closed 2020-01-31 by GilesBathgate