GilesBathgate/RapCAD #12: Why not using gmio to quickly import/export STL meshes?

HuguesDelorme, 2016-06-30:
Hello,

  Just to let you know about the existence of the gmio project.
  **gmio is a free and open-source C library** providing I/O support for the STL file format(more CAD formats in next releases).
- it's **fast**: gmio is x10 times faster to write ascii STL than Assimp, see [benchmarks](https://github.com/fougue/gmio/wiki/4.-Benchmarks)
- it's **feature-rich**: ascii/binary, little/big endian, format probing, infos retrieval, ... see [gmio_stl](http://www.fougue.pro/docs/gmio/0.2/group__gmio__stl.html#details)
- it's **reusable**: based on streams, seamless integration with other 3rd party libs, see [gmio_support](http://www.fougue.pro/docs/gmio/0.2/group__gmio__support.html#details)
- it's **portable**: C90 conformant but it can use recent C features if available

**RapCAD could benefit better STL format support with gmio**, just try it!
`>>` https://github.com/fougue/gmio
`>>` [v0.2.1 ZIP archive](https://github.com/fougue/gmio/archive/v0.2.1.zip) or `git clone https://github.com/fougue/gmio.git`
`>>` [Build instructions](https://github.com/fougue/gmio/wiki/2.-Build-instructions)
`>>` [Examples of use](https://github.com/fougue/gmio/wiki/3.-Getting-started)

kintel, 2016-06-30:
@HuguesDelorme To my knowledge CeCill-B isn't GPL compatible, due to the extended advertisement clause. Any chance to relicense under plain CeCill, plain GPL, or perhaps a simpler, 2-clause BSD license or smth. similar?

HuguesDelorme, 2016-06-30:
@kintel Thanks to pin that point, I have to check that with the CeCILL jurists. Another solution that comes to my mind is to make gmio available under both licenses (GPL and LGPL versions). As digia does for Qt.
I will let you know when this licensing issue is solved !

GilesBathgate, 2016-06-30:
@HuguesDelorme Dual license works for me.

HuguesDelorme, 2016-07-05:
@kintel I have just released v0.3 of gmio, which is relicensed under BSD 2-clause.

https://github.com/fougue/gmio/releases

GilesBathgate, 2017-12-09:
Since I need conversion from rational numbers (Gmpq) this libraries internal representation (which uses float) is not appropriate for me.

closed 2017-12-09 by GilesBathgate

HuguesDelorme, 2017-12-11:
@GilesBathgate The STL binary format uses 32bit floats for 3D coordinates. With the STL ASCII format you may possibly have doubles but I don't think this is common. So when you eg export to STL anyway at the end you will have to deal with float values or am I missing something ?