GilesBathgate/RapCAD #17: Add icon to app on macOS

ajmas, 2020-01-20:
The app generated on macOS is only using the generic icon and not the RapCAD icon.

 - The icon can be added per: https://doc.qt.io/qt-5/appicon.html
 - A `rapcad.icns` file will need to be generated, though this will require the addition of larger icons, per: https://stackoverflow.com/questions/12306223/how-to-manually-create-icns-files-using-iconutil

ajmas, 2020-01-31:
@GilesBathgate do you have original vector images for the icon?

GilesBathgate, 2020-01-31:
There is a big version of the icon in help/about:

[image]
Does that help?

ajmas, 2020-02-08:
Thanks. Will follow the instructions here for the icons:

https://elliotekj.com/2014/05/27/how-to-create-high-resolution-icns-files

Won't  include the @2x, since I am not sure about the finer details

GilesBathgate, 2020-02-08:
There is no need to duplicate the icon files

```
giles@BPEU0197:~/projects/rapcad/icons$ fdupes . -r
./rapcad-32x32.png                      
./Appicon.iconset/icon_32x32.png

./rapcad-16x16.png
./Appicon.iconset/icon_16x16.png

./rapcad-64x64.png
./Appicon.iconset/icon_64x64.png
```

You can just symlink them.

ajmas, 2020-02-08:
ok. which direction would you want the symlink to go?

./Appicon.iconset/icon_32x32.png -> ./icon_32x32.png

or 

./icon_32x32.png -> ./Appicon.iconset/icon_32x32.png

GilesBathgate, 2020-02-08:
e.g: `ln -s ../rapcad-32x32.png icon_32x32.png`

GilesBathgate, 2020-02-08:
Can you also paste a screenshot of what the icon looks like on your mac to give me an idea.

ajmas, 2020-02-08:
Here is the screenshot. Just trying to see if the iconutil likes symlinks

[image]

GilesBathgate, 2020-02-08:
>Just trying to see if the iconutil likes symlinks

If not do them the other way, not really fussed.

ajmas, 2020-02-08:
Ok, resolved. Let me know if those changes are fine?

GilesBathgate, 2020-02-08:
LGTM.

GilesBathgate, 2020-02-08:
hrm, on my machine the symlinks appear broken, this is due to the link using Camel case directory name `AppIcon.iconset`, but the folder being `Appicon.iconset`  you should be able to fix with `git mv Appicon.iconset AppIcon.iconset`

ajmas, 2020-02-08:
Just trying to fix this. hopefully the last push fixes this. missed that since my machine is case-insenstive

commit 09b02632 (2020-02-09): Fixes #17 Add app icon for macOS (#22)

* Replace duplicate icons with symlinks
* Renamed folder
closed 2020-02-09 by GilesBathgate