GilesBathgate/RapCAD #18: Is shape introspection possible?

etjones, 2020-01-20:
One problem I've run into with OpenSCAD is that I need to know or calculate explicitly where any given object lands. If I want to do something like "Attach object B to the center of the right face of  cube C", I need to calculate exactly where that face will be and how big it is. Depending on the situation that may be more or less manageable, but historically that's been about the point where I pick up a commercial CAD program. Does RapCAD have any way to analyze & introspect, say, the bounding box of objects?

GilesBathgate, 2020-01-20:
Well a slight improvement to openscad you can place one cube relative to another i.e `translate([0,5,0])cube([1,1,1])translate([0,0,5])cube([1,1,1]);` This idea was taken from a library called "relativity" which was designed for openscad.

Note that the above is different to:
```
translate([0,5,0])cube([1,1,1]);
translate([0,0,5])cube([1,1,1]);
```
which is not ralative the only difference is the `;` character, which could be confusing if you ommitted it accidentally :(

GilesBathgate, 2020-01-20:
perhaps `align()` could take multiple children, meaning align B (second child) to A (first child), or perhaps this should be a separate `attach()` module.

etjones, 2020-01-20:
That is clever! Cheers.

I'm not entirely sure what I'm wishing for, besides that sometimes in a pro CAD tool like SolidWorks I think "Well, why is this so hard, I just need to make a set of boxes in groups of two every 11 intervals" or whatever-- quick in CSG languages like RapCAD or OpenSCAD, more involved and less clear in graphical CAD programs.

At the same time, when I'm returning to an earlier SCAD project, I think "Well, why is this so hard, I just need to put a new element just coplanar to that one face on the upper right of this object" -- quick in a CAD GUI, more involved in a CSG language. It's good to have a couple more tools to accomplish that here. And to remember that no matter what septic tank you're standing on, the grass is always greener someplace else ;-)

closed 2020-01-20 by etjones