GilesBathgate/RapCAD #20: Segementation fault on 006_polygon.rcad related test, in Travis

ajmas, 2020-01-21:
When I submitted a PR for docs only change the build failed, due to the test for  006_polygon.rcad creating a segmentation fault on both Linux and macOS X targets.

see: https://travis-ci.org/GilesBathgate/RapCAD/builds/639611885?utm_source=github_status&utm_medium=notification

Linux:

```
Test #145: 002_polygon.rcad.............................................. Passed
Test #146: 003_polygon.rcad.............................................. Passed
Test #147: 004_polygon.rcad.............................................. Passed
Test #148: 005_polygon.rcad.............................................. Passed
Test #149: 006_polygon.rcad..............................................Segmentation fault (core dumped)
The command "if [ $TRAVIS_OS_NAME = osx ]; then ./rapcad.app/Contents/MacOS/rapcad -t test; else xvfb-run --server-args="-screen 0 1024x768x24" ./rapcad -t test; fi;" exited with 139.
```

macOS:

```
Test #145: 002_polygon.rcad.............................................. Passed
Test #146: 003_polygon.rcad.............................................. Passed
Test #147: 004_polygon.rcad.............................................. Passed
Test #148: 005_polygon.rcad.............................................. Passed
Test #149: 006_polygon.rcad............................................../Users/travis/.travis/functions: line 109: 22923 Segmentation fault: 11  ./rapcad.app/Contents/MacOS/rapcad -t test
The command "if [ $TRAVIS_OS_NAME = osx ]; then ./rapcad.app/Contents/MacOS/rapcad -t test; else xvfb-run --server-args="-screen 0 1024x768x24" ./rapcad -t test; fi;" exited with 139.

```

GilesBathgate, 2020-01-21:
Yeah it's odd. Previously the test passes: https://travis-ci.org/GilesBathgate/RapCAD/builds/600449886?utm_source=github_status&utm_medium=notification

ajmas, 2020-01-21:
Wanted to try to run the tests locally, using:

`./rapcad.app/Contents/MacOS/rapcad -t test `

but I get told `RapCAD: Unknown option 't'`.

GilesBathgate, 2020-01-21:
You will have to build with only CONFIG+=coverage or define USE_INTEGTEST

ajmas, 2020-01-21:
ok, seeing the same issue on my computer:

```
╰─➤  ./rapcad.app/Contents/MacOS/rapcad -t test                              
RapCAD 0.9.git.49ff5cab
Qt:	 5.13.2
CGAL:	 4.14
Boost:	 1.71.0
Test #000: 000_treeprinter............................................... Passed
Test #001: 001_callback.rcad............................................. Passed
Test #002: 001_equality.rcad............................................. Passed
Test #003: 002_equality.rcad............................................. Passed
Test #004: 003_equality.rcad............................................. Passed
Test #005: 004_equality.rcad............................................. Passed
Test #006: 001_variable.rcad............................................. Passed
Test #007: 002_variable.rcad............................................. Passed
Test #008: 003_variable.rcad............................................. Passed
Test #009: 004_variable.rcad............................................. Passed
Test #010: 001_value.rcad................................................ Passed
Test #011: 002_value.rcad................................................ Passed
Test #012: 003_value.rcad................................................ Passed
Test #013: 004_value.rcad................................................ Passed
Test #014: 005_value.rcad................................................ Passed
Test #015: 006_value.rcad................................................ Passed
Test #016: 001_operator.rcad............................................. Passed
Test #017: 002_operator.rcad............................................. Passed
Test #018: 003_operator.rcad............................................. Passed
Test #019: 004_operator.rcad............................................. Passed
Test #020: 005_operator.rcad............................................. Passed
Test #021: 006_operator.rcad.............................................[1]    66681 segmentation fault  ./rapcad.app/Contents/MacOS/rapcad -t test
```

GilesBathgate, 2020-01-21:
Thats not the same issue, its failing on a different test. Either way, It seems to be an issue with the latest version of cgal. I've tried disabling the problematic test and then the next one fails :(

ajmas, 2020-01-23:
If you can see what the seg fault is, maybe a ticket could be opened here:

https://github.com/CGAL/cgal

I'd do it myself, but I am don't know how to get the stack trace from a segmentation fault?

ajmas, 2020-01-23:
Actually, turns out on macOS a crash log is created in ~/Library/Logs/DiagnosticReports. I'll include the one I have, even it isn't quite the same scenario as on the build server.

[rapcad_2020-01-21-123149_Ghostwalker-Echo.crash.txt](https://github.com/GilesBathgate/RapCAD/files/4105274/rapcad_2020-01-21-123149_Ghostwalker-Echo.crash.txt)

GilesBathgate, 2020-01-24:
The segfault on your mac is happening in a different place to the one on the build server. You can probably verify that the test passes by running the following:

```
function test() {
a=2^8;
return a==256;
}

writeln(test());```

It needs more investigation before submitting a ticket to CGAL.

GilesBathgate, 2020-01-29:
This seems to only affect debug builds. I've temporarily made the builds on travis ci use release mode.

GilesBathgate, 2020-12-07:
For some reason re-ordering the headers in commit 4d4cbc1ad4da7f56ec1029f7b1bfef71e90cc87a solved the problem. I have re-enabled debug on travis ci.

closed 2020-12-07 by GilesBathgate