GilesBathgate/RapCAD #23: Updates and feature requests

WillAdams, 2020-03-05:
Please create /r/rapcad on reddit

Please add support for the Customizer

Please add support for a depth-mapped SVG export where the geometry is analyzed and each element is filled (say white is the top of the object, 50% a cut halfway through it, black is a perimeter/profile cut through the stock)

Please add a modular export format which would allow a naïve user to work up an export format --- I'd dearly love a way to export a .c2d file from it

GilesBathgate, 2020-03-05:
Great idea's @WillAdams But can you separate these into separate feature requests, and edit this one to be just the request to create `/r/rapcad`

Incidentally, I need to wait 30 days and have karma before I can create a subreddit.

WillAdams, 2020-03-05:
If you wish, I can create it, then add you as a moderator to /r/rapcad

GilesBathgate, 2020-03-05:
I should think that will work.

GilesBathgate, 2020-03-05:
Its u/GilesBathgate

WillAdams, 2020-03-06:
I've sent you an invite to be a moderator.

closed 2020-03-06 by GilesBathgate