GilesBathgate/RapCAD #25: Depth-map SVG support

WillAdams, 2020-03-05:
Please add support for a depth-mapped SVG export where the geometry is analyzed and each element is filled (say white is the top of the object, 50% a cut halfway through it, black is a perimeter/profile cut through the stock)

GilesBathgate, 2021-01-19:
@WillAdams would this be the same as for example using the `slice()` module at different heights within the model, and then exporting each slice to SVG, then finally combining the resulting SVG slices into a single svg which each slice set to a different color?

WillAdams, 2021-01-19:
Yes, that is exactly what I am describing.

In retrospect not sure if it's worth the effort --- there aren't any CAM tools which would directly make use of this, so one would still have to manually assign the tool paths, but the effective color-coding might be helpful to folks.

GilesBathgate, 2021-01-20:
@WillAdams Well I was thinking its a basic test output from the CAM output feature. I'll have a think as to how It can be implemented.

WillAdams, 2022-07-14:
For an example of this sort of thing, but w/ colour-coding, see:

https://github.com/crozone/openscad-trotec-export