GilesBathgate/RapCAD #26: Additional export formats

WillAdams, 2020-03-05:
Please add a modular export format which would allow a naïve user to work up an export format --- I'd dearly love a way to export a .c2d file from it

jwhevans, 2020-04-13:
I cannot find much info on .c2d files as they relate to 3D modeling.  What software are you using that reads/writes .c2d files?

GilesBathgate, 2020-04-13:
More info here: https://gilesbathgate.com/2013/02/01/site-downgrade/#comment-23587

WillAdams, 2020-04-13:
My apologies. .c2d files are the native format for Carbide Create, a closed-source (but freely available for 2.5D modeling) tool which uses a JSON dump as its file format. There's a Pro mode which has a free 1 yr. license available in consideration of the current global health crisis (see: the Carbide 3D community forums).

GilesBathgate, 2020-04-13:
Talking of 2.5D I totally missed the non-planar printing thing of yesteryear 

https://www.youtube.com/watch?v=gmePlcU0TRw

Does anyone know if anything more happened with this? (Perhaps we can move to a different thread since its off-topic.)

WillAdams, 2022-07-14:
.c2d files are now pretty much off the table, but if there's some other format(s) which could profitably be exported, that'd be great.

WillAdams, 2022-07-17:
The project most like to non-planer printing is probably fullgcodecontrol:

http://fullcontrolgcode.com/

(still waiting on them to release the Python version)

GilesBathgate, 2022-07-17:
@WillAdams Yep, I heard about this before. Thanks for providing a link.