GilesBathgate/RapCAD #28: Example of align() and glide() modules?

jwhevans, 2020-04-13:
I have been testing RapCAD recently.  I like the handling of parameters and variables, especially assignment within conditionals and loops. This is much better than OpenSCAD in my opinion.

The assign() and glide() modules seem like useful features.  However, I have been unable to make them work.  Does anyone have an example of their use?

GilesBathgate, 2020-04-13:
The align module takes a vector of values between -1 and 1, with 0 aligning to the center. So for example -1 will align to the bottom, and +1 will align to the top, and 0 will align center.

```scad
module thing()
{
	difference(){
		cube(10);
          cube(5);
	}
}

align([-1,0,1])thing();
```

The glide module will extrude a 3d solid along a 2d path:
```scad
glide(){
cube(2);
polyline([[0,0,0],[10,0,0],[10,10,0],[10,10,10],[0,0,0]]);
}```

jwhevans, 2020-04-13:
Thank you.  I was trying the top, bottom, north, south, east, west from the modules help file and was getting nowhere.  Are these vectors actually defined and functioning?

GilesBathgate, 2020-04-13:
That's an alternate syntax, but it requires boolean parameters:
`align(top=true,north=false)...`
I will update the documentation soon to make it clearer.

closed 2020-04-14 by GilesBathgate