GilesBathgate/RapCAD #30: Package in the Debian repository

vanillajonathan, 2020-08-04:
Package RapCAD in the Debian free software repository.

GilesBathgate, 2020-08-04:
Thanks for your request. If you can provide details as to how to host the package in the Debian free software repository, it will be easier for me to achieve.

vanillajonathan, 2020-08-04:
I've never done it myself, but it seems you have already done most of the work needed since you have a Debian package, now you need to use the `dupload` ([Debian](https://packages.debian.org/search?keywords=dupload), [Ubuntu](https://packages.ubuntu.com/focal/dupload)) tool.

The [Debian New Maintainers' Guide](https://www.debian.org/doc/manuals/maint-guide/) have a [Uploading the package](https://www.debian.org/doc/manuals/maint-guide/upload.en.html) chapter.

There is an IRC channel on #debian-devel at irc.debian.org for help. There is also a mailing list at debian-devel@lists.debian.org.