GilesBathgate/RapCAD #32: Allow RapCAD to write out text files?

WillAdams, 2020-08-11:
If it were possible in RapCAD to write out a text file one could directly create a G-Code file to cut out a design.

See: https://www.reddit.com/r/RapCAD/comments/i2dsk1/managed_to_model_a_finger_joint_box_with_relieved/

GilesBathgate, 2020-08-11:
I am not really sure what text output you desire, have you tried `export->CSG product`

WillAdams, 2020-08-11:
I would like the ability to write out coordinate information based on calculations which would allow a module to write out a G-Code command for making the equivalent cut as the module is making when called.

That way I could create a 3D model by calling a series of modules to cut away at it, while simultaneously writing out equivalent G-Code commands --- send the G-Code to a CNC machine and you'd then have a physical object.

See the module "radiuscut"

in https://community.carbide3d.com/uploads/short-url/cb4ilRtY2Hestsi1kbgdht9yM0Z.zip

GilesBathgate, 2020-08-11:
So you want the `writeln()` module to be output to file?

WillAdams, 2020-08-11:
Yes, I'd like a command which would write out to/append to a file, yes. --- something like having a command in the 

module radiuscut(beginx, beginy, endx, endy) module which had:

writeln("G0 X",beginx, " Y", beginy," Z",safeheight);
writenln("G1 Z",cutheight," F",plungerate);
writeln("G1 X", endx," Y",endy);
writenln("G0 Z",safeheight," F",plungerate);

where plungerate, safeheight, cutheight were all global variables in addition to the parameters used for the module.

WillAdams, 2020-08-20:
Complete proof of concept on this (albeit w/ hand-edited Console -> G-Code conversion) here:

https://community.carbide3d.com/t/doing-cam-in-openscad/25214

GilesBathgate, 2020-08-20:
I can easily add some kind of feature to allow you to output this to a file, but you can already do it:

`rapcad radiuscut.rcad -o filename.gcode`

Am I missing something?

WillAdams, 2020-08-20:
Should that work in Windows? Tried 

"C:\Program Files (x86)\RapCAD\rapcad.exe" test.rcad -o test.gcode

and didn't get anything --- while I'm willing to use the commandline, I worry other folks won't --- a menu entry or a preference:

[ ] write out only writeln text using file extension ___ 

would be great.

WillAdams, 2022-07-12:
Tried this w/ the new version w/ a file which had the content:

    cylinder(1,2,3,center=false);
    writeln("Text"){};

using the command:

[image]

but no "filename.nc" was made --- as noted above, the ideal would be a preference to always write out the file, or a language command to write it out.

GilesBathgate, 2022-07-12:
@WillAdams The `-o` option only takes filenames ending with certain extensions. so .amf/.3mf/.stl is what you would need. If you are trying to redirect text based output to a file try `-r example.nc`

GilesBathgate, 2022-07-12:
@WillAdams I'll see if I can debug this for you. I didn't implement the feature with closing #32 in mind, it just came about from a little refactoring work I was doing.

GilesBathgate, 2022-07-12:
@WillAdams I think the problem is your test.rcad contains extra braces, it should be:

```openscad
cylinder(1,2,3,center=false);
writeln("Text");
```

Then you can create the output with:
```cmd
"c:\Program Files\RapCAD\rapcad.exe" test.rcad -o null -r example.nc
```

this will produce the output in example.nc as follows:

```
Text
```

(Tested on windows 11.)

WillAdams, 2022-07-12:
Thanks! That worked! Next up, files w/ G-code!

I'd be even more grateful if you could add either a preference, or a command to cause the file to be written out --- it'd be a huge convenience and would save me a lot of clicking.

WillAdams, 2022-07-12:
Works on Mac OS X as well --- apparently there you need to include -o null to get the executable to quit.

Also posted a pair of announcements to /r/openscad and /r/rapcad on Reddit --- hopefully that will help.

GilesBathgate, 2022-07-12:
@WillAdams I'd be surprised if this isn't possible in OpenSCAD, I know nophead used text output for BOM generation in nopscadlib. You might have to use the openscad.com wrapper instead of openscad.exe on windows.

GilesBathgate, 2022-07-12:
@WillAdams Ah yes, so it's the `writeln()` and `write()` modules that simplify your workflow. I added those in 2013! 7afca8aa5135a3d9daa82be29096d7e27ebfb3e7

WillAdams, 2022-07-12:
Yes, but isn't the writing out to a file new in 1.0.2? I could never get it to work before.

Anyway, thanks, and looking forward to knocking out a file to radially flatten a surface this evening and then doing a test cut.

GilesBathgate, 2022-07-12:
@WillAdams It is new for Windows, but technically it's always been possible in linux. i.e: `./rapcad -o null > example.nc`

WillAdams, 2022-07-12:
Got it. Thanks for implementing for Windows! (and making the interface consistent --- that's important to me).

Which do you think is a better fit, a user preference, or a command to force writing out the file?

GilesBathgate, 2022-07-12:
@WillAdams I think a user preference, I don't see it as being part of the language spec. On the other hand other 'programming' languages do allow writing to file via a second parameter, so I might implement both in the long run. I'll implement a preference first.

WillAdams, 2022-07-12:
Thanks! I agree that it feels a lot better.

If you're inclined, it can be really useful to do string manipulation --- if this hadn't come through, I was seriously considering going back to doing all this in TeX:

https://tug.org/TUGboat/tb40-2/tb125adams-3d.pdf

GilesBathgate, 2022-07-12:
> If you're inclined, it can be really useful to do string manipulation 

You will have to be more specific, can you open a new Issue for this. Perhaps a regex function/module will be most useful.

WillAdams, 2022-07-13:
I'll write up something when I come to a point where I run into a limit and need something for it.

In the meanwhile, wrote up a bit on this at:

https://forum.makerforums.info/t/rapcad-1-0-2-just-released-openscad-alternative-can-write-out-files/85701/8

GilesBathgate, 2022-10-10:
Fixed as discussed above.

closed 2022-10-10 by GilesBathgate