GilesBathgate/RapCAD #34: The "latest installer" for windows is 0.9.8

YUKI2eN3e, 2020-10-01:
Both 0.9.9 and 1.0.0 are missing windows installers.
The "latest installer" for windows is repcad_0.9.8_setup.exe.

(Not sure if this technically classifies as an issue, but it is a bit confusing when the instructions say to download the "latest installer" just to find that the latest version has no windows installer.)

GilesBathgate, 2020-10-02:
Thanks @YUKI2eN3e I didn't know there was much demand for installers, so I haven't got around to building one for windows. I'll try to get one built soon.

GilesBathgate, 2020-10-24:
The windows installer has been uploaded to the releases page. I plan to use appveyor continuous deployment for the windows builds so hopefully the latest release installers will always be available when I make a release.

closed 2020-10-24 by GilesBathgate