GilesBathgate/RapCAD #35: Summary for curated list of code CAD

Irev-Dev, 2020-10-31:
Hey Giles 👋 ,

I finally got around to putting together a list of all of the code-CAD projects that I know of [here](https://kurthutten.com/blog/curated-code-cad), and also on [Github](https://github.com/Irev-Dev/curated-code-cad).

I've put a short summary of each one, but it's hard to do them all justice. for RapCAD I've only got:
Another project inspired by OpenScad.

If you would like to give me a summary of the project, and perhaps some points of how it differs from other similar projects would be happy to include it. I think max three paragraphs would be about right. Open a PR on reply to this issue and I'll include it.

GilesBathgate, 2020-11-01:
I consider the key differences to be procedural vs functional programming language style, (i.e variables can be modified) and the use of arbitrary precision arithmetic throughout (meaning there are no unexpected double/float rounding errors).

There is also a [feature matrix](https://github.com/GilesBathgate/RapCAD/blob/master/doc/feature_matrix.asciidoc)

Irev-Dev, 2020-11-01:
Thanks Giles.

closed 2020-11-01 by Irev-Dev

GilesBathgate, 2020-11-01:
@Irev-Dev Any chance you can change your screenshot. 

This is Slic3r

[image]

This is RapCAD:

[image]

Irev-Dev, 2020-11-02:
@GilesBathgate  Oh sorry,
Clearly was absent minded when I did that.
Should be fixed now.

Let me know if there's anything else I can doy.

Irev-Dev, 2021-01-01:
Me again.

I also want to add some of the best real-world examples of Code-CAD being used out in the wild. Do you know of any good projects using RapCAD that I can feature?

So far what I've found is:
- [Acedemics using it for reactor design](https://github.com/ukaea/paramak)
- [Lego compatible blocks](https://fxbricks.com/)
- [specialised in-house equipment for drones](https://twitter.com/robsmallshire/status/1343477584458346497)
- [Pruse 3d printer](https://github.com/prusa3d/Original-Prusa-i3)

GilesBathgate, 2021-01-15:
I don't know of anyone specifically using RapCAD for their project instead of Openscad. There is a list of some notable ones here: https://www.openscad.org/gallery.html