GilesBathgate/RapCAD #36: Add continuous integration

ajmas, 2020-12-06:
GitHub now provides a means to do continuous integration, by means of workflows, so it could be interesting to start using this for building the project. 

One explanation, targeting a macOS:

https://rhonabwy.com/2020/05/09/continuous-integration-with-github-actions-for-macos-and-ios-projects/

If anyone wants to take up the challenge, you are welcome.

GilesBathgate, 2020-12-06:
@ajmas I already build macOS releases using TravisCI

ajmas, 2020-12-06:
@GilesBathgate I hadn't noticed. I see the config file now. Is Windows and Linux target too?

GilesBathgate, 2020-12-07:
Windows releases are built using Appveyor, and Linux Releases are built using PPA. It might be worth streamlining this in future using Github CI, but it's not a priority at this time.

GilesBathgate, 2021-01-02:
This seems to be a good idea as travis-ci have changed their pricing model.

ajmas, 2021-01-02:
I did some work for macOS for another project, but unfortunately the resulting binaries appear corrupt and I haven’t had time to investigate why. I am sharing it here to see if it helps, despite the current state:

https://github.com/ajmas/Boxer/tree/workflow-tweaking

GilesBathgate, 2021-01-12:
Continuous integration is now active.

closed 2021-01-12 by GilesBathgate