GilesBathgate/RapCAD #37: Usage of 'volume();"

Mortu1412, 2021-01-17:
i am new to this tool, so i just tried this, because i need something that can create formula based 3d Models and also calculate the volume

module example004()
{
	difference() {
		cube(30, center = true);
		sphere(20);
	}
}
volume();
example004();

i get the Error "Warning: cannot find module 'volume'"

Help is really appreciated

GilesBathgate, 2021-01-17:
@Mortu1412 Auxilary modules don't produce any geometry so they are suffixed with $ i.e:

```module example004()
{
difference() {
cube(30, center = true);
sphere(20);
}
}
volume$();
example004();
```

Mortu1412, 2021-01-18:
Thanks alot! As a Last question, how to assign this output to a parameter or something similar. I need to create 3D-Structures with complex geometry and a specific Volume defined by one thickness d. I'm planning to describe these geometries dependend on the parameter d and let the programm run iterative till it finds the d for the expected Volume - i think this should be pretty doable using your program. 

Also is it possible to automatically export more than one figure to an obj or stl like this? I would like to run the programm once and get for example 5 obj files with diffrent specific Volumes. 

Thanks for your time!

GilesBathgate, 2021-01-18:
>how to assign this output to a parameter or something similar
under Linux/macos you can just parse the output of:
```
rapcad volume_test.rcad -o volume_test.stl
Volume: 2284.38321957741668306
Total compiling time: 0h 0m 2s 369ms.
Total export time: 0h 0m 0s 0ms.
```

Mortu1412, 2021-01-19:
Thanks for your Answer, i think i can figure the Rest out by myself :)

closed 2021-01-19 by Mortu1412