GilesBathgate/RapCAD #39: Should include work in directories other than the current one?

WillAdams, 2022-07-12:
It's not listed in:

file:///C:/Program%20Files/RapCAD/user_guide.html

but it highlights in the program:

[image]

I have it in:

C:\Users\willa\OneDrive\Documents\OpenSCAD\libraries

and tried:

C:\Users\willa\OneDrive\Documents\RapCAD\Libraries\gcodepreview.scad

but no joy, it seems to only search in the current directory.

But even if the file is there:

"C:\Users\willa\Desktop\gcodepreview.scad"

and called with:

    include <gcodepreview.scad>;

The error message is:

    Can't open input file 'C:/Users/willa/Desktop/gcodepreviewing.scad'
    Line 3: syntax error at character 1: ''
    Warning: no top level object.
    Я:

GilesBathgate, 2022-07-13:
@WillAdams the `\` character is an escape character, you will have to either use `\\` or `\` 
e.g: `C:\\Users\\willa\\OneDrive\\Documents\\OpenSCAD\\libraries`
or: `C:/Users/willa/OneDrive/Documents/OpenSCAD/libraries`

WillAdams, 2022-07-13:
Thanks! All fixed!

Would it be desirable to load from a fixed location such as Documents/RapCAD/Libraries  ?

GilesBathgate, 2022-07-14:
>load from a fixed location such as Documents/RapCAD/Libraries

@WillAdams Sorry I hit close too soon, please open a new issue for this anyway.

closed 2022-07-14 by GilesBathgate