GilesBathgate/RapCAD #41: writeln text from modules appears out of order

WillAdams, 2022-07-16:
If one has modules such as "setupcut" and "gcut" in https://github.com/WillAdams/gcodepreview/blob/main/gcodepreviewing.scad

and one calls them using a file such as:

    include <C:/Users/willa/OneDrive/Documents/RapCAD/Libraries/gcodepreview.scad>
    
    difference() {
      setupcut(9, 4, 1, "Top", "Center")
    
      gcut(0, 0, 0, 4.5, 2, -1, 390);
    
    }

the text appears out-of-order in the console:

    G1 X 0  Y 0 Z 0
    G1 X 4.5  Y 2 Z -1
    (STOCK/BLOCK,  9 ,  4 ,  1 )
    G90
    G21
    (Move to safe Z to avoid workholding)
    G53G0Z-5.000
    M05
    Total compiling time: 0h 0m 0s 20ms.
    Я:

and in the file.

GilesBathgate, 2022-07-16:
@WillAdams Well, it's actually a feature.

This allows you to position objects relatively. i.e:
```openscad
cube([1,1,1])
  translate([0,0,1]cube([2,2,2])
    translate([0,0,2])cube([3,3,3]);
```

I admit it probably doesn't make sense for writeln, I guess it could emit a warning.

```openscad
writeln("c") { 
   writeln("a");
   writeln("b");
}
```

WillAdams, 2022-07-16:
Oh. I would have put all those in groups using {} if writing things out by hand.

The warning makes a lot of sense --- feel free to close this ticket or leave it open as a reminder to implement that.

GilesBathgate, 2022-07-16:
> I would have put all those in groups using {} if writing things out by hand.

Yes this is probably clearer.
```openscad
cube([1,1,1]) {
  translate([0,0,1]cube([2,2,2]) {
    translate([0,0,2])cube([3,3,3]);
  }
}
```
But I have to support both for it to work consistently. The idea is inspired by so called `relativity.scad`: https://github.com/davidson16807/relativity.scad