GilesBathgate/RapCAD #42: writeln command adds space after text and before variable

WillAdams, 2022-07-16:
If you have a module:

    module toolchange(toolno) {
    writeln("(TOOL/MILL, ",toolno,")");
    	writeln("M6T",toolno);
    }

it adds a space after the quoted text and before the variable is inserted:

    (TOOL/MILL,  102 )
    M6T 102

GilesBathgate, 2022-07-16:
This is intentional and it is how other programming languages such as python implement this
https://trinket.io/python3/e2987f0bcc

Just use the following:
```openscad
module toolchange(toolno) {
writeln("(TOOL/MILL, " ~ str(toolno) ~ ")");
writeln("M6T " ~ str(toolno));
}

GilesBathgate, 2022-07-16:
The `str()` function also doesn't insert spaces so you could also use:
```openscad
module toolchange(toolno) {
writeln(str("(TOOL/MILL, ",toolno,")"));
writeln(str("M6T ",toolno));
}
```

WillAdams, 2022-07-16:
OIC, that makes sense.

Thanks!

closed 2022-07-16 by WillAdams

GilesBathgate, 2022-07-17:
>The concatenation command doesn't seem to work as expected

That's because you are concatenating a string and a number `"M6T" ~ 1` yeilds `1`
If you concatenate two strings it works `"M6T" ~ str(1)` yeilds `M6T1`

I think this is probably a bug though, I will have to see if there is a good reason for it not to always cast to strings during concatenation. Bear in mind concatenation is also supposed to work with arrays `[1,2,3] ~ [4]` yeilds `[1,2,3,4]`