GilesBathgate/RapCAD #45: Add support for OpenSCAD CLI options, esp. "viewall"

WillAdams, 2022-08-13:
Using the nifty new OpenSCAD Graph Editor: https://github.com/derkork/openscad-graph-editor

but it calls OpenSCAD using a command-line option "viewall" which results in an error when calling rapcad.exe:

[image]

List of the options at:

https://en.m.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment

Having a superset of the OpenSCAD options would help interoperability a lot.

GilesBathgate, 2022-08-13:
I can quickly add a viewall option so it doesn't error. Getting it to fit the view might take a little longer. Do you want me to just ward off the error or can you wait for the full feature?

WillAdams, 2022-08-14:
Just warding off the error would be great!

The full feature support would be gravy, but I can wait for that. 

Thanks!

GilesBathgate, 2022-08-14:
@WillAdams you can test a Windows snapshot if you like https://github.com/GilesBathgate/RapCAD/suites/7805119228/artifacts/329357650

WillAdams, 2022-09-06:
Thanks! This is working great!

[image]

GilesBathgate, 2022-10-10:
Fixed

closed 2022-10-10 by GilesBathgate