GilesBathgate/RapCAD #47: Examples to start with

AlexDep, 2022-10-27:
Hi,
Can we add examples in a dropdown menu ?
That could help with the new users.

Best,
Alex

GilesBathgate, 2022-10-27:
I have some examples here:

https://github.com/GilesBathgate/RapCAD/tree/master/examples

AlexDep, 2022-10-28:
It's less an effort to start with. Examples are good to see what is possible when using a new software, I usually start with changing the examples to start learning. 

I thought it could be a nice thing to add for user...

GilesBathgate, 2022-10-28:
I think something to insert the basic primitives, and/or templates would be better. 

```openscad
/** difference of two solids */
difference(){
    // primitive(); // basic solid
    // primitive(); // solid to remove
}
```