GilesBathgate/RapCAD #48: Integrate RapCad into BRL-CAD

SamtheTruck, 2023-07-07:
Now before you break into uncontrollable laughter at my foolishness, or audacity. I'm fully aware this "might" be the equivalent of me asking you to build a space shuttle in my backyard with bailing wire and two by fours. But...

BRL-CAD is this serious industrial strength program that some say is the oldest open source still maintained program. It has huge amounts of everything but, it doesn't have scripting like RapCAD. I think the only scripting it has is command line in various programming languages. I could be wrong. And it's missing STEP file support. Look at the comparison of features on this page.

https://appmus.com/vs/brl-cad-vs-openscad

and here's

https://brlcad.org/

What you've done is ambitious. I'm thinking it would be nice to have a full set of options in one large open source program. The military is the one who paid to write all of BRL-CAD and they use it still for all sorts of stuff. There are a lot of plug-ins for it. It does use "Constructive solid geometry", so it might not be a totally crazy idea. It would also give your program the vast features and thousands and thousands of man-hours invested into features already done.

Now I’m not telling you to do anything just maybe think about it and maybe some day you might see some way to make this happen without a lot of pain (right, sure, ok).

If this idea is silly, please ignore.

BTW if I go to this page

https://github.com/GilesBathgate/RapCAD/blob/master/doc/user_guide.asciidoc

and click on "RapCAD manual".

(https://github.com/GilesBathgate/RapCAD/blob/master/doc/manual.html)

It goes nowhere.

GilesBathgate, 2023-07-07:
@SamtheTruck Its not a bad idea. I just don't think I am going to be able to dedicate and time to do something like this any time soon. 

Regarding the manual url, it only works when the documentation is compiled, the asciidoc file becomes an html file.
The source for the manual is here: https://github.com/GilesBathgate/RapCAD/blob/master/doc/manual.asciidoc

SamtheTruck, 2023-07-08:
Thanks for your reply. It was just an idea. Who knows, if you don't ask...

And thanks for all the hard work you've done.

closed 2023-07-08 by SamtheTruck

GilesBathgate, 2023-07-08:
Maybe this will help:
https://brlcad.org/wiki/SGI_Cube